arpitjindal97/raspbian-recipes #1: [Question] Does the script have to be launched at each startup ?

mdemonceau, 2018-02-05:
Hi !

First of all thank's for these scripts, it's very useful ;-)

I was wondering if the script `wifi-to-eth-route.sh` have to be launched at each startup ?
Because you're saying to modify `/home/pi/.config/lxsession/LXDE-pi/autostart`, but my raspberry runs the raspbian-stretch-lite OS which doesn't include LXDE or any graphical interface.

I came from [your post on stackexchange](https://raspberrypi.stackexchange.com/a/50073) and already have configured a raspberry (raspbian-jessie-lite) thanks to it, but given that you posted an update for stretch I haven't tried the same configuration.
Can you explain why the latter doesn't work anymore with stretch ?

Thanks by advance for your response ! ;-)

arpitjindal97, 2018-02-06:
Try the below workaround. Keep the file at this location 

`/home/pi/wifi-to-eth-route.sh`

Open the file **/etc/rc.local**

`sudo nano /etc/rc.local`

Add the script command before `exit 0`

It should look like this.

```
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
  printf "My IP address is %s\n" "$_IP"
fi

bash /home/pi/wifi-to-eth-route.sh

exit 0
```

File **/etc/rc.local** is executed on boot by root user so putting the startup command there will do the job.

The older configuration that was working in Jessie won't work due to major changes introduced in Debian Stretch version. So, I have made these script to work dynamically without touching the dhcpcd config file.

mdemonceau, 2018-02-10:
Thank you, it's working like a charm ! :-)

closed 2018-02-15 by arpitjindal97