arpitjindal97/raspbian-recipes #2: Modify to use as Gateway

stevemac00, 2018-02-21:
Your wifi-to-eth-route.sh works well. Thank you for posting as iptables baffle me.

Would you consider modifications to a new script to provide the pi as a gateway?

Specifically, I'd like the pi3 wifi to connect and pass the gateway ip to the eth port to connect to a router (pfSense). One could then use a MiFi or any alternate wifi connection as a failover or load balancing connection to internet.

arpitjindal97, 2018-02-21:
You can't bridge wlan0 with eth0 (internet source is wlan0) even if the WiFi chip supports 4addr. 

Raspberry Pi 3 chip doesn't support 4addr but i have even tried it with a USB WiFi adapter which supports 4addr. A better explanation is [here](https://raspberrypi.stackexchange.com/questions/51057/raspberry-pi-3-model-b-wireless-bridge-to-ethernet#55731).

You can however do vice-versa, bridge eth0 with wlan0 (internet source eth0). Other devices will connect to RPi Hotspot and will get the IP address directly from router.

stevemac00, 2018-02-21:
Thanks. You're far more knowledgeable than I so I should take your advice and stop trying.
But then I read about a Layer 3 solution using proxy arp for this very purpose.
https://wiki.debian.org/BridgeNetworkConnectionsProxyArp
I get the Pi3 wlan0 connected to wifi hotspot and it's net address is also shown on eth0 but my pfSense doesn't get the local IP of the Pi so I'm stuck.

arpitjindal97, 2018-02-22:
So, I have made this small script to automate Proxy ARP. I have tried it on my Raspberry Pi. Hosts connected to RPi on Ethernet side are getting IP addresses .

Verify the steps you did manually with those in my script. 

I couldn't get `avahi-daemon` to get started on boot yet, I will fix that soon. Till then, try this and let me know if something is missing.

stevemac00, 2018-02-22:
I appreciate your effort. I couldn't get it to work. I have internet on pi but host cannot connect on eth0 as it cannot get an IP address. I attached the script output as well as ifconfig.
[out.txt](https://github.com/arpitjindal97/raspbian-recipes/files/1749581/out.txt)

arpitjindal97, 2018-02-23:
The problem is on this line
```
dhcp-helper: cannot bind DHCP server socket: Address already in use
```

`dhcp-helper` uses UDP Port 67, it looks like some other process is already using this port before `dhcp-helper` is started. 

You can list all the port being used and their respective program name with pid.
```
sudo netstat -tulpn
```
Find the process name and kill it, you must have installed some package which is using this port. 
Try it on **fresh** Raspbian Stretch Image and it will work fine.

stevemac00, 2018-02-23:
I see what you're referring to about dhcp-helper port in use. That was from dnsmasq in earlier scripts so I did fresh install of 2017-11-29-raspbian-stretch-desktop.img
sudo apt-get install parprouted dhcp-helper
configured wpa-supplicant.conf
reboot - wifi connects fine
run wifi-to-eth-bridge.sh
As before, ifconfig shows wifi inet address but the eth0 attached host gets a link address of 169.254.117.137
I tried everything I could. (I also noticed after a minute of so, the mdns relinqushed the addresses.)

arpitjindal97, 2018-02-23:
`dhcp-helper` is responsible for assigning IP addresses to devices connected to `eth0`. 

`mdns` is related to `avahi-daemon`, we don't need it as of now. Follow these steps 

- Comment the last line which will start `avahi-daemon` in the script. 
- Run the script
- type this command manually, it should execute successfully

      sudo /usr/sbin/dhcp-helper -r /var/run/dhcp-helper.pid -b wlan0 

Now check if LAN device is getting IP address or not.

closed 2018-06-04 by arpitjindal97

arpitjindal97, 2018-06-26:
I was trying this script with my another router and found that it wasn't working.

Same script working with one router but not with another. There must be some sort of security in router firmware which doesn't let Wi-Fi to bridge.