arpitjindal97/raspbian-recipes #4: How to disable?

Noirad, 2018-03-20:
Hey, thanks for sharing this script and posting on this forum (https://raspberrypi.stackexchange.com/questions/48307/sharing-the-pis-wifi-connection-through-the-ethernet-port)!

At the moment, I’m trying to disable the feature to use Ethernet normally. 

Could you offer some advice about how to do so?

Any feedback would be greatly appreciated! 

Thanks so much!

arpitjindal97, 2018-03-21:
Please explain what do you actually want to do ? How do you want to use Ethernet ?

Noirad, 2018-03-24:
Hey @arpitjindal97, that is exactly what I'd like to do!

Noirad, 2018-03-24:
Thanks for the swift reply, didn't see your notification!

arpitjindal97, 2018-03-25:
I don't understand your question properly. Please explain a bit more. Are you trying to do exact same thing what the script does manually ? If yes, then you can just read the content of the script and understand what's going on under the hood.

Noirad, 2018-03-25:
I'm trying to stop the script and connect my raspberry pi to the internet via Ethernet.

It looks like during boot the script is launched and the IP tables are forwarded.

If I remove the script from /home/pi and reboot, will I be able to connect my raspberry pi normally to Ethernet again?

I would simply this out, but I'm working on this unit remotely. I do have someone who can connect the Ethernet cable so I can confirm.

arpitjindal97, 2018-03-25:
Yes, you can remove the script from `/home/pi` or just rename it and you will be able to use ethernet normally.

Noirad, 2018-03-25:
Perfect, I'll give it a try today!

arpitjindal97, 2018-04-18:
Is your problem solved now ?

closed 2018-06-04 by arpitjindal97