arpitjindal97/raspbian-recipes #6: problem with bind-interfaces in /etc/dnsmasq.conf

korreckj328, 2018-04-13:
Ok I'm beating my head against the wall with this one so I figured I'd post an issue to see if I can figure it out.

I've used your script in the past and everything worked great,  it made it a lot easier than doing everything manually.

However I needed to make up a new sd card and I hadn't backed up....  So I'm starting from the beginning again unfortunately.

I'm using the aiyprojects-2018-2-21 img which is a google aiy kit version of raspbian stretch.
dnsmasq is already installed,  and the system is fully updated.

When your script runs I get Job for dnsmasq.service failed because the control process exited with error code.

checking systemctl status dnsmasq shows:

pi@raspberrypi:~ $ systemctl status dnsmasq
● dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
   Loaded: loaded (/lib/systemd/system/dnsmasq.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2018-04-13 08:28:53 EST; 14s ago
  Process: 4695 ExecStop=/etc/init.d/dnsmasq systemd-stop-resolvconf (code=exited, status=0/SUCCESS)
  Process: 1171 ExecStartPost=/etc/init.d/dnsmasq systemd-start-resolvconf (code=exited, status=0/SUCCESS)
  Process: 4760 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=2)
  Process: 4757 ExecStartPre=/usr/sbin/dnsmasq --test (code=exited, status=0/SUCCESS)
 Main PID: 1170 (code=exited, status=0/SUCCESS)

Apr 13 08:28:53 raspberrypi systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Apr 13 08:28:53 raspberrypi dnsmasq[4757]: dnsmasq: syntax check OK.
Apr 13 08:28:53 raspberrypi dnsmasq[4760]: dnsmasq: unknown interface usb0
Apr 13 08:28:53 raspberrypi systemd[1]: dnsmasq.service: Control process exited, code=exited status=2
Apr 13 08:28:53 raspberrypi systemd[1]: Failed to start dnsmasq - A lightweight DHCP and caching DNS server.
Apr 13 08:28:53 raspberrypi systemd[1]: dnsmasq.service: Unit entered failed state.
Apr 13 08:28:53 raspberrypi systemd[1]: dnsmasq.service: Failed with result 'exit-code'.

If I remove the bind-interfaces command from /etc/dnsmasq.conf,  dnsmasq will start so I know the problem has something to do with that line.  (but it doesn't actually bind the interfaces)

Now,  my first thought was that I had something wrong with my config and that I needed to edit your script to use usb0 for something.  (Your script is unchanged at this point)

However ifconfig produces:

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.2.1  netmask 255.255.255.0  broadcast 192.168.2.255
        inet6 fe80::a99c:7101:9e08:45b1  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:03:81:34  txqueuelen 1000  (Ethernet)
        RX packets 142  bytes 28674 (28.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 163  bytes 31199 (30.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 20  bytes 1557 (1.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 20  bytes 1557 (1.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.0.93  netmask 255.255.255.0  broadcast 10.0.0.255
        inet6 2601:40e:4000:cdc7::ce6c  prefixlen 128  scopeid 0x0<global>
        inet6 fe80::4e91:97fb:4096:81aa  prefixlen 64  scopeid 0x20<link>
        inet6 2601:40e:4000:cdc7:60e5:f552:85c5:e62d  prefixlen 64  scopeid 0x0<global>
        ether b8:27:eb:56:d4:61  txqueuelen 1000  (Ethernet)
        RX packets 1172  bytes 145291 (141.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 372  bytes 63341 (61.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

So I'm really unsure why it's trying to do something with usb0

arpitjindal97, 2018-04-13:
If I am guessing right than you are using `wifi-to-eth-route.sh` script. Looking at the logs of `dnsmasq` tells that you have edited the script and replaced `eth0` with `usb0`. 

If you have nothing to do with usb interface than undo any changes you made to script or download it again. Also explain a little bit about how to want to route the traffic.

Feel free to ask anything if it doesn't work.

korreckj328, 2018-04-13:
Yeah,  I'm sorry I am using wifi-to-eth-route.sh,  I'm just using it to route from the wifi to ethernet so that an old computer that doesn't have a working wifi card anymore can use it to get on my home network.  

As far as me editing your script,  I haven't.  That's why I'm scratching my head.  This is a direct copy and paste from the copy of your script that I'm using.  I was hoping you would have some idea where usb0 is coming from.

#!/bin/bash

# Share Wifi with Eth device
#
#
# This script is created to work with Raspbian Stretch
# but it can be used with most of the distributions
# by making few changes. 
#
# Make sure you have already installed `dnsmasq`
# Please modify the variables according to your need
# Don't forget to change the name of network interface
# Check them with `ifconfig`

ip_address="192.168.2.1"
netmask="255.255.255.0"
dhcp_range_start="192.168.2.2"
dhcp_range_end="192.168.2.100"
dhcp_time="12h"
eth="eth0"
wlan="wlan0"

sudo iptables -F
sudo iptables -t nat -F
sudo iptables -t nat -A POSTROUTING -o $wlan -j MASQUERADE  
sudo iptables -A FORWARD -i $wlan -o $eth -m state --state RELATED,ESTABLISHED -j ACCEPT  
sudo iptables -A FORWARD -i $eth -o $wlan -j ACCEPT 

sudo sh -c "echo 1 > /proc/sys/net/ipv4/ip_forward"

sudo ifconfig $eth $ip_address netmask $netmask

# Remove default route created by dhcpcd
sudo ip route del 0/0 dev $eth &> /dev/null

sudo systemctl stop dnsmasq

echo -e "interface=$eth\n\
bind-interfaces\n\
server=8.8.8.8\n\
domain-needed\n\
bogus-priv\n\
dhcp-range=$dhcp_range_start,$dhcp_range_end,$dhcp_time" > /etc/dnsmasq.conf

# This file will work if resolveconf package is installed
# Due to new updates in dnsmasq
cp /etc/dnsmasq.conf /run/dnsmasq/resolv.conf

sudo systemctl start dnsmasq

arpitjindal97, 2018-04-13:
Try this workaround to debug the issue 

- Remove script command from autostart, let pi boot normally
- Attach monitor, keyboard to it and run the script normally
- Open another terminal, and check if dnsmasq is running or not.
- If it is not then open the file and see if it contains `eth0` or `usb0`
- if it is `usb0` the edit it and replace it with `eth0`
- Start the dnsmasq service manually
- check again the status of dnsmasq if it failed or running

If it is running then restart the pi and start script again manually and check if everything is fine or not.

If still it is not working then you should try it in official raspbian image.

korreckj328, 2018-04-15:
Yeah works on official raspbian.  Just doesn't on the AIY-images.

arpitjindal97, 2018-04-15:
Google's official [website](https://aiyprojects.withgoogle.com/vision/) says AIY image is for Raspberry Pi Zero. The device doesn't have ethernet port and maybe you are trying this image on RPi 3. So, dnsmasq which is already installed is searching for usb0 (hardcoded).

korreckj328, 2018-04-15:
Double checking myself that I pulled the right image,  as I'm using the image from here:  https://aiyprojects.withgoogle.com/voice/

The voice kit can work with the 0 with limited functionality but is meant to be on a 3.

arpitjindal97, 2018-04-15:
So, I will try AIY voice image and see if I can do something :thinking:

arpitjindal97, 2018-04-15:
Solved the problem, turns out to be google devs have put `/etc/dnsmasq.d/googadget-dnsmasq.conf` file.

Dnsmasq first reads `/etc/dnsmasq.conf` and then reads all the config files in `/etc/dnsmasq.d/` directory. So, `googadget-dnsmasq.conf` file was overriding the configurations.

I have made a change in script now. Config will now be stored in `/etc/dnsmasq.d` directory and all other files will be removed from there.

korreckj328, 2018-04-15:
Yup,  it all works now.  I figured it'd be something like that,  just ended up beyond my know-how of linux to find what they changed.  Thank you very much.

closed 2018-04-15 by korreckj328