arpitjindal97/raspbian-recipes #7: Using with Proxy

edelciomolina, 2018-04-13:
Is possible to use with Proxy Squid?

I have my squid proxy configurations working fine, I know because when I configure my PC to use proxy (through wlan0) they deny the URLs.

But, when I connect my PC to eth0 (that is sharing my connection), all is allow. Is there a way to block all and allow only if I set the proxy?

arpitjindal97, 2018-04-13:
I have never used `squid-proxy` but I will definitely help you. Share your squid configs to help me get started. 

I will try to make your current configuration work i.e eth0 shared internet should also be passed through squid.

It should work and then we can make a script for raspberry which will do all the configuration at run time and others users will be needed to connect to RPi via ethernet or use ip address of RPi as proxy server

edelciomolina, 2018-04-13:
@arpitjindal97 thank you so much!

**MY SQUID.CONF**
```
http_port 3128 transparent
  
auth_param basic program /usr/lib/squid/ncsa_auth /etc/squid/passwd
auth_param basic children 5
auth_param basic realm Molina
auth_param basic credentialsttl 2 hour
auth_param basic casesensitive off
 
logfile_rotate 6
cache_mem 64 MB
maximum_object_size_in_memory 64 KB
maximum_object_size 700 MB
minimum_object_size 0 KB
cache_swap_low 90
cache_swap_high 95
cache_dir ufs /var/spool/squid 2048 16 256
cache_access_log /var/log/squid/access.log
refresh_pattern ^ftp: 15 20% 2280
refresh_pattern ^gopher: 15 0% 2280
refresh_pattern . 15 20% 2280

acl all src 0.0.0.0/0.0.0.0
acl manager proto cache_object
acl localhost src 127.0.0.1/255.255.255.255
acl SSL_ports port 443 563
acl Safe_ports port 21 80 443 563 70 210 280 488 59 777 901 1025-65535
acl purge method PURGE
acl CONNECT method CONNECT
acl liberados url_regex "/etc/squid/liberados"
acl redelocal src 192.168.0.0/24
acl ncsa_users proxy_auth REQUIRED

http_access deny manager
http_access allow purge localhost
http_access deny purge
http_access deny !Safe_ports
http_access deny CONNECT !SSL_ports
http_access allow localhost

http_access allow ncsa_users liberados
http_access deny redelocal
http_access deny all
```
... and some other files, if help.

**MY INTERFACES**
```
auto lo
iface lo inet loopback
 
auto eth0
allow-hotplug eth0
iface eth0 inet static  
    address 192.168.2.1
    netmask 255.255.255.0
    network 192.168.2.0
    broadcast 192.168.2.255
 
allow-hotplug wlan0
iface wlan0 inet manual
wpa-roam /etc/wpa_supplicant/wpa_supplicant.conf
iface default inet dhcp
```
**MY DNSMASQ.CONF**
```
interface=eth0      # Use interface eth0  
listen-address=192.168.2.1 # listen on  
# Bind to the interface to make sure we aren't sending things 
# elsewhere  
bind-interfaces
server=8.8.8.8       # Forward DNS requests to Google DNS  
domain-needed        # Don't forward short names  
# Never forward addresses in the non-routed address spaces.
bogus-priv
# Assign IP addresses between 192.168.2.2 and 192.168.2.100 with a
# 12 hour lease time
dhcp-range=192.168.2.2,192.168.2.100,12h
```

edelciomolina, 2018-04-13:
@arpitjindal97 I forgot to mention that I tried to put that line right after your iptables commands:
`sudo iptables -t nat -A PREROUTING -i $eth -p tcp --dport 80 -j REDIRECT --to-port 3128`

But doesn´t work.

arpitjindal97, 2018-04-13:
Remove the existing rules from script and try these iptables 

```
sudo iptables -t nat -A PREROUTING -i eth0 -p tcp -- dport 22 -j REDIRECT -- to-ports 22
sudo iptables -t nat -A PREROUTING -i eth0 -p udp -- dport 53 -j REDIRECT -- to-ports 53
sudo iptables -t nat -A PREROUTING -i eth0 -p tcp -- syn -j REDIRECT -- to-ports 3128
```

arpitjindal97, 2018-04-13:
I haven't been able to run squid properly :smile: 

It prompted many error, I managed to get rid of those error and at the end it looks like I was able to run it but i think it is still not configured.

**/usr/lib/squid/ncsa_auth**
```
auth_param basic program /usr/lib/squid/ncsa_auth /etc/squid/passwd
auth_param basic children 5
auth_param basic realm Molina
auth_param basic credentialsttl 2 hour
auth_param basic casesensitive off
```

**/etc/squid/liberados**
```
^http(s)?://([a-zA-Z]+).google.com/.*$
```

**squid.conf** is same. I also set these 2 iptables rules from [here](https://wiki.archlinux.org/index.php/Squid#Transparent_web_proxy). 

replaced the `SQUIDIP` with the IP assigned to me by router.
Then opened firefox and used localhost:3128 as proxy for all protocols and it can still access opens google.com

edelciomolina, 2018-04-16:
@arpitjindal97 Your suggestion works like a charm!! Thank you man! 
You save my day!

[image]

closed 2018-04-16 by edelciomolina