arpitjindal97/raspbian-recipes #15: Use with resinOS

jhabdas, 2018-06-08:
I think this should be possible but I'm not sure. Basically I want to run the `wifi-to-eth` script on a machine running resinOS. Thoughts?

https://forums.resin.io/t/route-wireless-to-wired-with-resinos-2-x-and-rpi-3/3219/4

arpitjindal97, 2018-06-08:
I tried to understand the resinOS and it's pretty much complex. The file system is read-only. 
Everything is done via container. 

So, I have created docker container for that and successfully routed the Wi-Fi to ethernet. 

There were some changes required to make it work inside container.

jhabdas, 2018-06-11:
Okay, this works perfectly. It was easy and can be repeated very easily now that it's containerized. Thanks so much!! Want a pull to add this to the README?

closed 2018-06-11 by jhabdas

arpitjindal97, 2018-06-11:
Added you as collaborator. It would be great to help in documentation