arpitjindal97/raspbian-recipes #16: isolate eth0 from the main network

saikhurana, 2018-06-22:
hey arpit,
i really appreciate the work that you have done and the code works really well. i don't know if i am missing out something  but here's what i want.
1. Raspberry Pi should connect to the main network of my university and provide me with a ethernet. but here's something that i want in addition. i want the rpi to act as a insulator between network. i want the pi to assign the ip address.

also is there a way we can configure rpi to automatically login to the captive portal of my collage.

arpitjindal97, 2018-06-22:
I see what you are referring to is Bridge Wi-Fi to Ethernet. 
See #2 . 

Layer 2 bridging is not possible. Try the bright [script](wifi-to-eth-bridge.sh) for layer 3 solution. 

> also is there a way we can configure rpi to automatically login to the captive portal of my collage.

That's what i wasn't able to do with my college Wi-Fi. Provide the steps of how you connect with your college Wi-Fi. Let's see if i can do any help

saikhurana, 2018-06-22:
i am a total noob when it comes to networking and how things are. i am learning along the way. about the captive portal, my collage is yet to start and so will tell you once i get there in mid august.
also can we connect via social media so it would be easier to conversation.

arpitjindal97, 2018-06-22:
You can connect with me on [keybase](https://keybase.io/arpitjindal97) and [telegram](https://t.me/arpitjindal97)

closed 2018-06-23 by saikhurana