arpitjindal97/raspbian-recipes #18: script request for wifi-to-wifi& eth.

saikhurana, 2018-06-30:
hey, 
buddy you are doing an awesome job and these scripts are very helpful. i want to request for a script that can take wifi from one interface and share the internet using another interface as AP and at the same time sharing the internet via the ethernet interface as well.

arpitjindal97, 2018-07-09:
Take wifi-to-eth-route script. Make sure this script is working correctly on Pi. 

Then open wifi-to-wifi-route script and try to understand each step. Add required lines from this script to above working script. Leave dnsmasq part for now and add all relevant things one by one. Keep testing each merge by doing a reboot of Pi. 

Do it till your Access Point is visible. Devices won't get IP from Access Point. 

Now, edit the `dnsmasq` part. You will have to google it. Configure it so that single instance of `dnsmasq` will provide dhcp service on both `eth0` and `wlan1`. Both these interfaces must have same IP i.e. 192.168.2.1

arpitjindal97, 2018-07-09:
Closing it for now. Re-open it any time if you need any help.

closed 2018-07-09 by arpitjindal97