arpitjindal97/raspbian-recipes #19: using the scripts with pihole

saikhurana, 2018-07-07:
these scripts are really helpful but is there a way we can use these scripts with pihole. Pihole has a built in DHCP server so is there a way we can disable the DHCP server on these scripts so the pihole provides the ip address

arpitjindal97, 2018-07-09:
DHCP service is provided by `dnsmasq`. You can comment the lines dealing with `dnsmasq`. You have to configure pihole's `dnsmasq` to listen on your specified interfaces also.

closed 2018-07-09 by arpitjindal97

arpitjindal97, 2018-07-09:
[Here](https://www.reddit.com/r/pihole/comments/7y1ake/pihole_setup_help_for_a_usb_dongle/) is a thread on reddit which can help you. It looks like, so many people are trying to make it work with pihole. I have been busy working on other project, I will take out some time for pihole