arpitjindal97/raspbian-recipes #20: wifi to eth not working

gitkav, 2018-07-17:
Hi Arpit 

I have rapberry VERSION="9 (stretch)"
I tried your script wifi-to-eth-route.sh, but unfortunately its not working for me.
The device that wants to use the raspberry wifi connection is a TV.
I have connected the TV to raspberry pi through LAN cable and ran you script on reboot.
My network interface file is

My /etc/wpa_supplicant/wpa_supplicant.conf has
country=GB
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
network={
    ssid="Home WiFi"
    psk="password_goes_here"
}

Can you please let me know what I am missing

arpitjindal97, 2018-07-17:
Thanks for notifying. I will fix the README with correct config.

closed 2018-07-17 by arpitjindal97

gitkav, 2018-07-18:
Hi Arpit,

With the above changes also it is not working.

gitkav, 2018-07-18:
Its working now, the date and time on my tv was incorrect because of which TV was rejecting the TLS certificate. Thanks

commit 26d6211f (2018-07-17): Fixed #20 updated wpa config