arpitjindal97/raspbian-recipes #22: wifi-to-wifi Hotspot Working but no Internet

YisroelTech, 2018-08-20:
I ran the wifi-to-wifi-route script and the hotspot was successfully created. I'm able to connect devices to it, but they get no internet connection (my USB adaptor is connected to internet and the Pi has internet.)

What can it be?

arpitjindal97, 2018-08-20:
I will debug the script by today

YisroelTech, 2018-08-20:
Thanks, let me know if you need anything from me to help (I'm a raspberry pi noob, but can follow instructions.)

YisroelTech, 2018-08-20:
So sorry...

I Now figured out the issue... (my ignorance...) namely that the USB WiFi adoptor was at some point changed names from wlan1 to wlx00c0ca9202dd! So I changed the name in the script, rebooted, and it works just great!

Thanks a lot

closed 2018-08-20 by YisroelTech