arpitjindal97/raspbian-recipes #23: wifi-to-eth-bridge.sh unable to get ip from DHCP

gareks, 2018-09-12:
I has been following your work on wifi-to-eth-bridge.sh over Raspberry.

I followed your guide with 2018-06-27-raspbian-stretch.img.

However, no matter what kind of method, I cannot get the Client PC connect to Raspberry eth0 to get an IP from DHCP.

If I use static IP in the client PC, I can get the internet access through the eth0 port.

Thanks in-advance for your help.

arpitjindal97, 2018-09-12:
only DHCP is not working right ? PC is getting internet with static IP but not from DHCP. 

Give me some time, I will try to debug the issue. This is Layer 3 solution actually, so it doesn't work with most of the Routers. Maybe you can try doing it with your Mobile Hotspot.

gareks, 2018-09-12:
Yes, thx for the quick response Sir.
The PC can get on internet with static IP I set which is the same network of Raspberry wlan0
But using DHCP, I cannot get the ip assigned from DHCP server.
I tried also with iPhone Personal Hotspot and not working.

gareks, 2018-09-12:
Actually, I would like to explain better my project to you. I'm using Raspberry as Desktop computer and try to use the eth0 port of Raspberry for my VoIP SIP Phone.
At the beginning, I tried to use wifi-to-eth-router script, However VoIP SIP always having trouble behind NAT, that require the router enable SIP passthrough features which doesn't work well with your script.

Thats why I turned to use your wifi-to-eth-bridge script which allow the eth0 device obtain a standalone IP on the network, and I tested it, that can solve all the issue when I plug a VoIP SIP phone to eth0, however the last problem is my device cannot obtain IP from DHCP, I can only put a static IP to the VoIP phone and connect it to Raspberry eth0

arpitjindal97, 2018-09-17:
Maybe I should leave this issue for some more time and you will find the solutions yourself :laughing:

gareks, 2018-09-17:
Yes exactly 
I found the issue already and its related to dhcp relay 

Anyway 
My last issue is just the wifi-to-eth-bridge.sh cannot work with rc.local 

But wifi-to-eth-route.sh is okay

arpitjindal97, 2018-09-17:
See this file `~/.config/Lxde/lxsession/autostart` directory names may vary

saikhurana, 2018-09-17:
the correct location is 

> /home/pi/.config/lxsession/LXDE-pi/autostart

gareks, 2018-09-18:
However my project required to auto login to a non sudor user 
And your solution won't work for mine

arpitjindal97, 2018-09-18:
Have a look at #14 workaround also

closed 2018-10-09 by arpitjindal97