arpitjindal97/raspbian-recipes #29: Use device on eth0

stuxxix, 2019-01-19:
the script works very well, thank you.
i mean it realy work for the regular usage imo. i can surf with all listed (later in this text) devices in the internet.

i got the following setup (each number represent a device and them are connected):

(1) ISP/Modem/Router (fritzbox 6490, LAN+WIFI)
(2) pi1 with latest raspbian lite and your script 
(3) for example a lokal-webserver
(4) My PC

(2) connected with wifi to (1)
(3) connected with eth to (2)
(4) connected with eth to (1)

Now i try with (4) to open a website on (3) which wont work because diffrent subnets as far i know

maybe it is possible to reach 4 with 3 and vice versa.

saikhurana, 2019-01-19:
wifi-to-eth-bridge should work because then the pi would not be assigning the ip's. Your router(1) will assign the ip address to the device connected to the ethernet port of the raspberry pi.

arpitjindal97, 2019-01-20:
(3) is in different subnet created by (2). One workaround you can do is port forwarding in (2). Then you will be able to access webserver at specific port of (2).

closed 2019-03-13 by arpitjindal97