arpitjindal97/raspbian-recipes #30: Unable to reach 192.168.2.x subnet - incorrect routing?

clickworkorange, 2019-02-17:
I have a Pi (Raspbian Stretch) running `wifi-to-eth-route.sh`. It is connected to my NAS via Ethernet, with the Pi in turn connecting to my WWAN router (172.28.172.1) via WiFi. I can access the Pi on the IP address it gets from my WWAN router (172.28.172.3), but I cannot access the NAS (192.168.2.9) from the Pi. 

`$ ip address
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:04:f5:1b brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.1/24 brd 192.168.2.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet 169.254.97.91/16 brd 169.254.255.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::1647:d1a3:ace6:1d5b/64 scope link 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:51:a0:4e brd ff:ff:ff:ff:ff:ff
    inet 172.28.172.3/24 brd 172.28.172.255 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::b6be:bf19:8b0e:eeb2/64 scope link 
       valid_lft forever preferred_lft forever`

`$ ip route show
default via 172.28.172.1 dev wlan0 src 172.28.172.3 metric 303 
169.254.0.0/16 dev eth0 proto kernel scope link src 169.254.97.91 metric 202 
172.28.172.0/24 dev wlan0 proto kernel scope link src 172.28.172.3 metric 303 
192.168.2.0/24 dev eth0 proto kernel scope link src 192.168.2.1`

`$ cat /proc/sys/net/ipv4/ip_forward
1`

`$ sudo iptables --table nat --list POSTROUTING --verbose
Chain POSTROUTING (policy ACCEPT 129 packets, 11026 bytes)
 pkts bytes target     prot opt in     out     source               destination         
   28  3210 MASQUERADE  all  --  any    wlan0   anywhere             anywhere`

`$ ping 192.168.2.9
PING 192.168.2.9 (192.168.2.9) 56(84) bytes of data.
^C
--- 192.168.2.9 ping statistics ---
26 packets transmitted, 0 received, 100% packet loss, time 25919ms`

Could this be because the default route points to my WWAN router?

arpitjindal97, 2019-02-18:
This problem is strange

Below line means any traffic of `192.168.2.xxx` will be passed via eth0

`192.168.2.0/24 dev eth0 proto kernel scope link src 192.168.2.1` 

So, configuration is absolutely correct. Try to ping `192.168.2.1` from NAS. Maybe it is blocking pings

clickworkorange, 2019-02-19:
Thanks. As so often is the case, this turned into yet another opportunity for the palm of my hand to get intimate with my face; the firewall on my NAS had autoblocked the Pi's IP :) Nothing to see here... move along...

closed 2019-02-21 by arpitjindal97