arpitjindal97/raspbian-recipes #31: How configure second wlan?

rexy666, 2019-03-24:
How can I configure a second wireless adapter, I tried using network/interfaces but it just shows wlanX not associated upon reboot

arpitjindal97, 2019-03-25:
Do one thing, before running any script just connect the usb wifi normally and try to connect it to your home wifi. Check if it getting connected and obtaining the IP address. 

Also, append the output of `ifconfig` before and after plugging the external usb wifi. 

Is it usb wifi or data dongle with sim card ?

rexy666, 2019-04-01:
Sorry, you can close this thread. The issues had nothing to do with your script. Your script is amazing. Thanks for thaking the time of coding it and supporting it

closed 2019-04-02 by arpitjindal97