arpitjindal97/raspbian-recipes #33: wifi-to-eth-bridge DNS server

Byterapper, 2019-04-18:
Hi,
the wifi-to-eth-bridge.sh is working for me, but my windows PC on eth0 wants to use the bridge's IP as a DNS server. 
If I manually set a DNS server on the PC, it is able to resolve addresses, the raspberry is able to resolve without any further settings.

Is there a way to have the raspberry resolve addresses for the PC as well?

arpitjindal97, 2019-04-22:
I think everything is working fine according to the setup. PC on eth0 should have IP of Raspberry as DNS server.

I actually don't understand which addresses you want Pi to resolve.

Byterapper, 2019-04-23:
Of course, I'll try:
as written, PC on eth0 is able to use the bridge properly, both to local devices (on same wireless LAN as raspberry) as well as the internet.
The PC also gets a valid IP in the subnet and can be reached from other local devices on the wireless.
It also sets (unless I set DNS manually to a static address of course) the DNS server to the raspberry's IP.
with the raspberry IP set, the name resolution of e.g. google.com is not working on the eth0 PC, while on the raspberry it is working (if I ping the name e.g. via terminal). So the raspberry is itself capable of doing so.

My Laptop ,directly connected to the WiFi, is using the wireless router to resolve addresses. If I enter manually this router's IP or any other DNS server like 8.8.8.8 on the PC behind eth0, it also resolves URLs.

So I see two possible solutions if I don't want to manually assign DNS after all:
a) the raspberry needs some extra configuration (how?) to announce it's own DNS server to the eth0 PC. so in my case the WiFi router's address. Ideally this also would happen automatically, but I also would be happy about setting this statically on raspberry side.
b) the raspberry itself needs some extra configuration (how?) or tool (which?) to be able to resolve incoming eth0 DNS requests by itself, requesting that info from its own DNS server and passing it on to eth0.

I hope this explains my thoughts better! :)

arpitjindal97, 2019-05-14:
Are thoses packages installed in raspberry pi which are mentioned in begining of script ?

Byterapper, 2019-05-14:
> Are thoses packages installed in raspberry pi which are mentioned in begining of script ?

Yes, avahi-daemon, parprouted and dhcp-helper are installed

arpitjindal97, 2019-05-14:
Try this, instead of wifi router use your mobile hotspot. Connect RPi to it and see if PC is getting DNS automatically or not.

closed 2019-07-31 by arpitjindal97