arpitjindal97/raspbian-recipes #36: wifi-to-eth-bridge.sh on RPi4 for Verizon Hot Spot to Wifi Router

toqana, 2019-12-29:
I live in a rural area and my primary Internet Service Provider goes down often.

The wifi-to-eth-bridge.sh worked well to connect the Wifi hotspot from my Verizon cell phone to a WiFi router using a RPi4 as the bridge.

The WiFi router setting for "Route Modem IP" must be changed to match the IP address assigned by the Verizon hot spot.

The result is that all my devices have Internet access again through the same Wifi router with the same subnet addresses.

Thank you a very helpful script.

arpitjindal97, 2019-12-29:
I didn't get the issue

toqana, 2019-12-29:
There is no problem. This was a comment to say the script worked fine on a RPi4 and allowed me to connect a cell phone WiFi hot spot to a WiFi router ethernet input port Thank you.

closed 2019-12-29 by toqana