arpitjindal97/raspbian-recipes #37: /home/pi/.config/lxsession/LXDE-pi/autostart not present in ubuntu buster

Schizo, 2020-01-30:
Hi, 

I was trying to follow the walkthrough, however, figured that the folder doesn't exist for me:

```
    /home/pi/.config/lxsession/LXDE-pi/autostart
```

I'm not too familiar with raspbian, but is there another way adding it to autostart in `ubuntu buster`?

arpitjindal97, 2020-01-31:
The path of this autostart script depends on the Desktop Environment. In Raspbian, LXDE comes default, so above path is specific to LXDE.

Based on your DE you are using in ubuntu buster. you can configure it.

Schizo, 2020-02-01:
Sorry, that's my bad I accidentally typed `ubuntu buster` instead of `raspbian buster`. Anyhow I finally switched to the lite version. Will see how it goes from there.

Schizo, 2020-02-09:
I've used the lite version and it worked like a charm.

closed 2020-02-09 by Schizo