arpitjindal97/raspbian-recipes #40: SSH to client of wifi-to-eth-route.sh

mariomadproductions, 2020-03-25:
I've connected another machine to the network using wifi-to-eth-route.sh. How can I ssh into it? The IP address according to ``hostname -I`` on the client machine is 127.0.0.1, when I was expecting an 192.168.0.X address for sshing into.

arpitjindal97, 2020-03-25:
@saikhurana

saikhurana98, 2020-03-25:
@mariomadproductions There are 2 ways to go about this problem. 
1. Create a port forward. This is the proper way to go about solving the issue. You would be required to make some changes in dnsmasq configuration and possibly IPTABLES. Since the  device youre trying to access in on a subnet. So you raspberry pi or whatever device you're using to run the script is a router which appears on your parent network. So the router should make provisions to allow that feature.
2. The easiest way is to just ssh to your raspberry pi or whatever the device is you're using to run the script. Once you're ssh'd you can access any device on that subnet. So you can just ssh to you device that is connected via ethernet.

mariomadproductions, 2020-03-25:
Actually I'm using the Rasperry Pi as a "workstation" so I only need to be able to access the client machine from that. But the problem is the ip address of the client according to the client itself is ``127.0.0.1``, which I doubt is an IP I can use to ssh into the client.

mariomadproductions, 2020-03-25:
Seems like I was just using the wrong command. ``ip route get 1`` print info which includes the correct IP address.

closed 2020-03-25 by mariomadproductions