arpitjindal97/raspbian-recipes #41: Bridge using a vpn

MarcoMihaiCondrache, 2020-04-08:
Hello man! Your script works fine but i have a little problem.
I use openvpn to create a vpn connection to a provider.
When i create this connection a new interface is added but the internet connection stops working.
I think that happens because there is no bridge between the eth0 and the new interface.
How can i bridge the eth0 to the vpn interface and the new interface to the wlan0? Thank you for the help

arpitjindal97, 2020-04-08:
Which interface is source is source of internet?

cc @saikhurana98

MarcoMihaiCondrache, 2020-04-08:
The internet source is wlan0

saikhurana98, 2020-04-08:
I'm not entirely sure but just use the virtual interface created by vpn as a source for your vpn'd internet.

MarcoMihaiCondrache, 2020-04-08:
> I'm not entirely sure but just use the virtual interface created by vpn as a source for your vpn'd internet. 

Thank you I will try.

MarcoMihaiCondrache, 2020-04-09:
> I'm not entirely sure but just use the virtual interface created by vpn as a source for your vpn'd internet. 

I tried it by changing the wlan variable inside the script with the name of the new interface created but the problem of internet absence has not been solved

arpitjindal97, 2020-04-09:
You need to create 2 bridges, one is wlan0 to vpn0 and vpn0 to eth0

MarcoMihaiCondrache, 2020-04-09:
> You need to create 2 bridges:
> 
> 
> 
>  - wlan0 to vpn0
> 
>  - vpn0 to eth0

if I'm not mistaken and from what I read online the bridge between tun0 and wlan0 is made when the vpn connection is created then the two interfaces are already connected.  So now, as you said, I miss the bridge between tun0 and eth0.  What do I have to do to do this?

MarcoMihaiCondrache, 2020-04-09:
Another thing, tun0 has a metric of 0 so all internet traffic is directed to that interface

arpitjindal97, 2020-04-14:
you can look at code in script, where bridging is done, just try to copy the same code paste it same place just below existing and replace the interfaces

MarcoMihaiCondrache, 2020-04-15:
@arpitjindal97 ok I’ve resolved with your method

closed 2020-04-15 by MarcoMihaiCondrache