arpitjindal97/raspbian-recipes #42: Port Forwarding

Nickjallday, 2020-04-23:
So first off thank you so much for this script. It's really awesome I use the "wifi-to-eth-route.sh" on one of my 3d printers that only has an ethernet interface.

My issue is for this script to work well for me I need to be able to forward port 80 from an ethernet IP in my case (192.168.2.2). I accomplished this on my 3d printer by adding variables:

duet2port="80"
duet2ip="192.168.2.2"

and the following two lines after "sudo iptables -A FORWARD -i $eth -o $wlan -j ACCEPT"

sudo iptables -t nat -A PREROUTING -p tcp -i $wlan --dport $duet2port -j DNAT --to-destination $duet2ip:$duet2port

sudo iptables -A FORWARD -p tcp -d $duet2ip --dport $duet2port -m state --state NEW,ESTABLISHED,RELATED -j ACCEPT

For some reason on this second setup the wifi-to-eth-route.sh works I can access the internet from behind the pi ethernet, but I cannot get access 192.168.2.2 port 80 from the wifi network. I know it's not necessarily an issue, but I was hoping for a bit of guidance on how to accomplish this on my second pi setup. The only differences in the setups are this one is a Pi3 running Buster. The Pi2 is running Jesse.

arpitjindal97, 2020-04-24:
@saikhurana98

Nickjallday, 2020-05-02:
Any idea? I haven't made any progress.

arpitjindal97, 2020-05-02:
Your iptables looks fine to me. Can u try SSHing into RPi and do curl over `192.168.2.2`

Nickjallday, 2020-05-08:
I figured it out, I had to uncomment net.ipv4.ip_forward=1 in /etc/sysctl.conf. Thanks guys!

closed 2020-05-08 by arpitjindal97