arpitjindal97/raspbian-recipes #43: No longer working

jaylfc, 2020-05-05:
Hi,

I love these scripts and have used them for a while now. Since the Raspbian update they no longer work. I tried changing my default by to iptables legacy and unmasked dnsqmask and the script now runs without errors but doesn’t actually work.

Any ideas?

Thanks in advance!

arpitjindal97, 2020-05-05:
I really haven't got time to test script recently

jaylfc, 2020-05-05:
I wish I knew enough to fix it, I’ll keep playing around, if I manage anything I’ll post on here!

albertfaromatics, 2020-05-22:
Any news on this? Got the same problem!! Script work fine. It runs without errors, but the devices I connect to the ethernet are simply not detected nor assigned an IP

arpitjindal97, 2020-05-22:
I do not RPi right now with me.

Can u guys try to execute the steps in scripts 1 by 1 in console and see which lines throws error ?

albertfaromatics, 2020-05-22:
Just run the code of wifi-to-eth-route 1 by 1, no errors. I can check in ifconfig that eth0 has a correct IP of 192.168.2.1, with correct mask and gate

Connected my PC to the ethernet of the raspberry. Nothing happens

albertfaromatics, 2020-05-22:
Ok now it worked. I just rebooted the RPI and now it worked. I need to do some more testing I think

jaylfc, 2020-05-22:
I never got it to function correctly due to the change of networking system in the latest version. I manually created a bridge following Debian guides in the end.

albertfaromatics, 2020-05-22:
Ok so after some more testing:
- If I add the script to /etc/rc.local, to run at boot, I get the following:
    · ifconfig configured properly. with IP and mask.
    · I can see the DHCPREQUEST on syslog of my computer
May 22 16:27:27 raspberrypi dnsmasq-dhcp[803]: DHCPREQUEST(eth0) 192.168.2.36 30:9c:23:17:fe:17
May 22 16:27:27 raspberrypi dnsmasq-dhcp[803]: DHCPACK(eth0) 192.168.2.36 30:9c:23:17:fe:17 DESKTOP-L671BNR
    · My computer gets IP, but no internet connection, and I cannot ping it from the Raspberry
- If I DO NOT add the script to run at boot to /etc/rc.local
    · ifconfig OK again as before, with correct 
    · same as before, DHCPREQUEST on syslog
    · My computer gets IP AND Internet connection. I can ping and ssh from Raspberry

albertfaromatics, 2020-05-22:
> I never got it to function correctly due to the change of networking system in the latest version. I manually created a bridge following Debian guides in the end.

Do you have the script? I'm struggling to create it.....

arpitjindal97, 2020-05-22:
@jaylfc Can u provide clean steps here ?, we can add that to README

jaylfc, 2020-05-22:
I’m out right now, I will have to find the page I used in my history. I will post as soon as back :)

jaylfc, 2020-05-22:
This is the page I used I think:

https://raspberrypi.stackexchange.com/questions/88214/setting-up-a-raspberry-pi-as-an-access-point-the-easy-way

albertfaromatics, 2020-05-22:
I think you are using eth-to-wifi, no? So you are getting internet to the rpi from the ethernet and creating a WiFi AP.

I need the other way around: I have internet on WIFI and I want to connect my computer to the ethernet port of the RPI

jaylfc, 2020-05-22:
Nope, I have my pi sharing incoming from wlan0 to Ethernet. I plugged a router into it to create a new separate network and all devices have internet and local connectivity.

albertfaromatics, 2020-05-25:
Still not working for me.

I've seen that if I run the script WITHOUT anything connected to the ethernet, and then I connect a switch to the port, then it works

If the script is run with something already connected to the ethernet port of the raspberry, it fails

jaylfc, 2020-05-25:
Ok, give me a few hours, I just realised I could open my bash history file and see exactly what I did to get it working. I will post later :)

jaylfc, 2020-05-26:
_**Make sure to:

update-alternatives --config iptables

Change to legacy.**_

The dnsmasq service may need unmasking after installation.

Here is what I did, I have pasted the entire file under each nano command, you should be able to just replace unless you have already made some critical changes to your configs.

####start#### 

sudo apt-get install dnsmasq
sudo nano /etc/dhcpcd.conf

######start of file#####

interface eth0
static ip_address=192.168.220.1/24
static routers=192.168.220.0

hostname

# Use the hardware address of the interface for the Client ID.
clientid

# Persist interface configuration when dhcpcd exits.
persistent

# Rapid commit support.
# Safe to enable by default because it requires the equivalent option set
# on the server to actually work.
option rapid_commit

# A list of options to request from the DHCP server.
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes
# Respect the network MTU. This is applied to DHCP routes.
option interface_mtu

# A ServerID is required by RFC2131.
require dhcp_server_identifier

# Generate SLAAC address using the Hardware Address of the interface
#slaac hwaddr
# OR generate Stable Private IPv6 Addresses based from the DUID
slaac private

######end of file######

sudo service dhcpcd restart
sudo mv /etc/dnsmasq.conf /etc/dnsmasq.conf.orig
sudo nano /etc/dnsmasq.conf

#####start of file####

interface=eth0 # Use interface eth0 
listen-address=192.168.220.1
# Specify the address to listen on 
# Bind to the interface 
server=8.8.8.8 # Use Google DNS domain-needed
# Don'tforward short names 
bogus-priv # Drop the non-routed address spaces.
dhcp-range=192.168.220.50,192.168.220.150,12h # IP range and lease time

####end of file####

sudo nano /etc/sysctl.conf

####start of file####

net.ipv4.ip_forward=1

####end of file####

sudo sh -c "echo 1 > /proc/sys/net/ipv4/ip_forward"
sudo iptables -t nat -A POSTROUTING -o wlan0 -j MASQUERADE 
sudo reboot
sudo systemctl restart dnsmasq.service 
sudo systemctl daemon-reload
sudo sh -c "iptables-save > /etc/iptables.ipv4.nat"
sudo nano /etc/rc.local

####start of file####

#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

# Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
printf "My IP address is %s\n" "$_IP"
fi

iptables-restore < /etc/iptables.ipv4.nat

exit 0

####end of file####

sudo nano /etc/iptables.ipv4.nat 

####mine looks like this####
# Generated by iptables-save v1.8.2 on Sun May 17 21:34:45 2020
*filter
:INPUT ACCEPT [8429:5899839]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [7033:785423]
-A FORWARD -i wlan0 -o eth0 -m state --state RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -i eth0 -o wlan0 -j ACCEPT
COMMIT
# Completed on Sun May 17 21:34:45 2020
# Generated by iptables-save v1.8.2 on Sun May 17 21:34:45 2020
*nat
:PREROUTING ACCEPT [183:33480]
:INPUT ACCEPT [183:33480]
:OUTPUT ACCEPT [197:14662]
:POSTROUTING ACCEPT [20:2492]
-A POSTROUTING -o wlan0 -j MASQUERADE
COMMIT
# Completed on Sun May 17 21:34:45 2020

####end of file####

sudo systemctl restart dnsmasq.service 
sudo systemctl daemon-reload

####end###

If its not working maybe reboot.

Hope this helps!

jaylfc, 2020-05-26:
_**Make sure to:

update-alternatives --config iptables

Change to legacy.**_

The dnsmasq service may need unmasking after installation.

Here is what I did, I have pasted the entire file under each nano command, you should be able to just replace unless you have already made some critical changes to your configs.

####start####

sudo apt-get install dnsmasq
sudo nano /etc/dhcpcd.conf

######start of file#####

interface eth0
static ip_address=192.168.220.1/24
static routers=192.168.220.0

hostname

Use the hardware address of the interface for the Client ID.
clientid

Persist interface configuration when dhcpcd exits.
persistent

Rapid commit support.
Safe to enable by default because it requires the equivalent option set
on the server to actually work.
option rapid_commit

A list of options to request from the DHCP server.
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes

Respect the network MTU. This is applied to DHCP routes.
option interface_mtu

A ServerID is required by RFC2131.
require dhcp_server_identifier

Generate SLAAC address using the Hardware Address of the interface
#slaac hwaddr

OR generate Stable Private IPv6 Addresses based from the DUID
slaac private

######end of file######

sudo service dhcpcd restart
sudo mv /etc/dnsmasq.conf /etc/dnsmasq.conf.orig
sudo nano /etc/dnsmasq.conf

#####start of file####

interface=eth0 # Use interface eth0
listen-address=192.168.220.1

Specify the address to listen on
Bind to the interface
server=8.8.8.8 # Use Google DNS domain-needed

Don'tforward short names
bogus-priv # Drop the non-routed address spaces.
dhcp-range=192.168.220.50,192.168.220.150,12h # IP range and lease time

####end of file####

sudo nano /etc/sysctl.conf

####start of file####

net.ipv4.ip_forward=1

####end of file####

sudo sh -c "echo 1 > /proc/sys/net/ipv4/ip_forward"
sudo iptables -t nat -A POSTROUTING -o wlan0 -j MASQUERADE
sudo reboot
sudo systemctl restart dnsmasq.service
sudo systemctl daemon-reload
sudo sh -c "iptables-save > /etc/iptables.ipv4.nat"
sudo nano /etc/rc.local

####start of file####

#!/bin/sh -e

rc.local
This script is executed at the end of each multiuser runlevel.
Make sure that the script will "exit 0" on success or any other
value on error.
In order to enable or disable this script just change the execution
bits.
By default this script does nothing.
Print the IP address
_IP=$(hostname -I) || true
if [ "$_IP" ]; then
printf "My IP address is %s\n" "$_IP"
fi

iptables-restore < /etc/iptables.ipv4.nat

exit 0

####end of file####

sudo nano /etc/iptables.ipv4.nat

####mine looks like this####

Generated by iptables-save v1.8.2 on Sun May 17 21:34:45 2020
*filter
:INPUT ACCEPT [8429:5899839]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [7033:785423]
-A FORWARD -i wlan0 -o eth0 -m state --state RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -i eth0 -o wlan0 -j ACCEPT
COMMIT

Completed on Sun May 17 21:34:45 2020
Generated by iptables-save v1.8.2 on Sun May 17 21:34:45 2020
*nat
:PREROUTING ACCEPT [183:33480]
:INPUT ACCEPT [183:33480]
:OUTPUT ACCEPT [197:14662]
:POSTROUTING ACCEPT [20:2492]
-A POSTROUTING -o wlan0 -j MASQUERADE
COMMIT

Completed on Sun May 17 21:34:45 2020
####end of file####

sudo systemctl restart dnsmasq.service
sudo systemctl daemon-reload

####end###

If its not working maybe reboot.

Hope this helps!

Vincent14, 2021-03-08:
Hi everyone, is your problem solved?

I seem to meet the same problem and created a PR for a fix I would offer you to test (if applicable) https://github.com/arpitjindal97/raspbian-recipes/pull/50

Any help is welcome! =)

vincentkoevoets, 2021-05-11:
Well I don't think the problem is solved.. I found out that the script still only works when there is nothing connected to the ethernet-port. I have to boot the Pi, execute the script, and only then connect a device to the ethernet port, otherwise it will not work. Devices will get an ip-address, but always state 'no internet connection'. I don't know how to help directly, but I'm loving the script so I'm hoping we can solve this thing together.