arpitjindal97/raspbian-recipes #46: wifi-to-eth-bridge.sh works but only if the ethernet client uses a static IP

dhry, 2020-09-20:
I'm using the wifi-to-eth-bridge.sh script, after basically a full day of trying to figure out how to share my wireless connection across wifi. After multiple solutions it's basically only thing that has worked. Except..

If my PC that is connected to the Pi via ethernet has automatically assign IP and DNS server, it does NOT get an autoassigned IP address, even though I'm using my router as the gateway in the settings, ie 192.168.0.1

If I force any static IP address on my subnet for the PC to use, ie 192.168.0.63, mask 255.255.255.0, gateway 192.168.0.1 and DNS server eg 1.1.1.1, then it works and I have internet access.

Part of the bridge script setup involves installing dhcp-helper. This always fails to install and run, stating that it cannot bind to DHCP address, likely because dhcpcd is running? I don't know if this is significant.

How do I set up the bridge script so that the client hanging off of the Pi's ethernet will get an auto-assigned IP address instead of needing to force static?

arpitjindal97, 2020-09-20:
Try this once:

 - Comment line https://github.com/arpitjindal97/raspbian-recipes/blob/master/wifi-to-eth-bridge.sh#L59
 - Run the script
 - Run the `dhcp-helper` command manually
 - if it says, it is already running, then do `ps aux | grep dhcp-helper` and kill the process
 - Run the `dhcp-helper` command manually

dhry, 2020-09-20:
Does not work. I see that the kill command you suggest is already in line 40 of the script. It was running already. I killed the proc, restarted it, it looked to have started fine but my device is still not autoacquiring it's own IP from DHCP. No Internet access on the ethernetted PC. Tried the Windows troubleshooter on the PC, it reset the adapter, says adapter does not have a valid IP configuration. But all I have is the ipv4 properties set to acquire an IP and DNS server automatically.

Script output as follows:

```
pi@pi3b-bridge:~ $ sudo bash wifi-to-eth-bridge.sh
Killing parprouted
Stopping dhcp-helper service
Killing dhcp-helper
Stopping avahi-daemon service
Warning: Stopping avahi-daemon.service, but it can still be activated by:
  avahi-daemon.socket
Killing avahi-daemon
Creating temp avahi conf file
Starting parprouted ...
Starting dhcp-helper ...
dhcp-helper: no process found
Flushing eth0 IP addr
Assigning IP to eth0 from wlan0
Starting avahi-daemon ...
Process 18040 died: No such process; trying to remove PID file. (/run/avahi-daemon//pid)
Found user 'avahi' (UID 108) and group 'avahi' (GID 113).
Successfully dropped root privileges.
avahi-daemon 0.7 starting up.
Successfully called chroot().
Successfully dropped remaining capabilities.
No service file found in /etc/avahi/services.
Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::13a4:dae7:c108:d4c9.
New relevant interface wlan0.IPv6 for mDNS.
Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.0.62.
New relevant interface wlan0.IPv4 for mDNS.
Joining mDNS multicast group on interface eth0.IPv4 with address 192.168.0.62.
New relevant interface eth0.IPv4 for mDNS.
Network interface enumeration completed.
Registering new address record for fe80::13a4:dae7:c108:d4c9 on wlan0.*.
Registering new address record for 192.168.0.62 on wlan0.IPv4.
Registering new address record for 192.168.0.62 on eth0.IPv4.
Server startup complete. Host name is pi3b-bridge.local. Local service cookie is 3582646508.
```

I even added the following to the script:
```
echo "Starting dhcp-helper ..."
sudo killall dhcp-helper
sudo /usr/sbin/dhcp-helper -r /var/run/dhcp-helper.pid -b $wlan &
```

Still nada. This is crazy.

arpitjindal97, 2020-09-21:
Can you try replacing `$wlan` with `$eth` ? I'm doing a wild guess here :)

swr85260, 2020-11-22:
I'm a padawan, but after a couple of hours, I may have found the hint :

1/ As your comment from Sept 21st, in the "wifi-to-eth-bridge.sh" file
echo "Starting dhcp-helper ..."
sudo /usr/sbin/dhcp-helper -r /var/run/dhcp-helper.pid -b $eth &

2/ in the "dnsmasq.conf" file, uncomment an update 
dhcp-range=192.168.1.50,192.168.1.150,255.255.255.0,12h

It works, but may overlap with existing IP
Appreciate your review