arpitjindal97/raspbian-recipes #47: Terminal Echo Overflow

myxxmikeyxx, 2020-12-23:
Would it be possible to not echo everything to terminal and maybe to a log file? or do something like > /dev/null
I do not really care about the output and could do it to a file but I
[image]
 hate it echoing to the terminal and causing this 
(look at attached IMG)

myxxmikeyxx, 2020-12-23:
I added > /dev/null 2>&1 to the rc.local file and it seems to work, but for some reason when I first connect to it after a reboot it moves the cursor and start at the front and writes over the pi@raspberry:~ $   part but it still works fine after that and doesn't show anything.

This happens whenever I have a device try to connect or reconnect or disconnect, any fix? It's not a super big deal just a minor thing that annoys me but I can just hit enter and made do.

Video Link (Too Large to attach): https://drive.google.com/file/d/1s3Y3FkPeVytMuPPkVrCybiLmIy_HimOb/view?usp=sharing

arpitjindal97, 2020-12-25:
I guess, you will have to make same modification at this line

https://github.com/arpitjindal97/raspbian-recipes/blob/master/eth-to-wifi-route.sh#L71

myxxmikeyxx, 2021-02-20:
Sorry forgot to close this. But yes I’d you just tell it to put the whole files output or individual line output to /dev/null then it seemed to stop that problem for me

closed 2021-02-20 by myxxmikeyxx