arpitjindal97/raspbian-recipes #49: wifi-to-eth-route.sh  doesn't work on latest Raspberry Pi OS

Vincent14, 2021-03-06:
It seems to configure everything OK, and run without any error, but my client eth computer isn't able to join internet, has only received an IP on 192.168.2.x

myxxmikeyxx, 2021-03-06:
Please see if this still works and send a screenshot or copy of the code so I can know if I'm testing the correct version.
`cat /etc/os-release`
[Resource](https://pimylifeup.com/raspbian-version/)

Vincent14, 2021-03-06:
I can see "Raspbian GNU/Linux 10 (buster)"

I used this link https://github.com/arpitjindal97/raspbian-recipes/raw/master/wifi-to-eth-route.sh

myxxmikeyxx, 2021-03-06:
Okay what version?
Also with that script did you preinstall 'dnsmasq'?
if not disable the script conect to wifi and then do `sudo apt-get update && sudo apt-get install dnsmasq`

Vincent14, 2021-03-06:
Hem what version of which component? ^^' 

Yes I've installed dnsmasq previously and configured wpa_supplicant without raspi-config (works fine, used for apt update)

myxxmikeyxx, 2021-03-06:
Have you don’t anything else or just make a new SD card with that img and tried this? If it’s new then I will try to flash a new one and do the same thing you have. Also have you done `sudo apt-get update && sudo apt-get upgrade`? If so I will do that on mine and see if I’m getting the same problem

Vincent14, 2021-03-06:
I have to confess that I tried another tutorial around dnsmasq/iptables previously but I was ~sure to reset any parameter before trying your script, and I'm okay to reformat everything if necessessary :/ I didn't update/upgrade so i'm currently doing this first

Vincent14, 2021-03-06:
I use a Raspberry Pi 1 B and a D-link DWA-140 wifi dongle (rt2800 driver)

myxxmikeyxx, 2021-03-06:
Well the wifi adaptor should not matter for this, as long as it connects to the wifi fine. It only would matter is using the script to make it a wifi hot spot. I will have to do some testing tonight, I have a raspberry pi 3b+
also if you are just testing install of stuff I recommend flashing PINN.
(https://github.com/procount/pinn)
It’s like a better NOOBS and has a lot more options.

myxxmikeyxx, 2021-03-07:
I did just try it with a fresh install and it worked fine.  All I did was do wget <link>
then chmod 777 <script>
then sudo bash <script>
plugged the Ethernet into my laptop and turned the wifi off on my laptop and it worked. 
I did turn thw wifi off on my leptop and made sure it had the ethernet icon.
Also I did use a crossover ethernet cable but i do not think that matters, but I do not know for sure.

Vincent14, 2021-03-07:
Oops sorry, just understood one of your previous questions, i use Raspbian GNU/Linux 10 (buster) LITE ^^" The one without desktop environment. 

I'm currently: setup a new SD card with Raspbian GNU/Linux 10 (buster) LITE, used raspi-config to setup wifi, apt update/upgrade and will retry your script on top of this. I only have "straight-through" cables :/

myxxmikeyxx, 2021-03-07:
Like I said I don’t think that matters but I do not know for sure. I know patch cables are used for device to device communication without a switch. 
Do you have a small network switch that you can try and see if it is because the cable isn’t crossover? 
I will have to look around and see if I have a regular Ethernet cable and try that and see if it still works fine.

Vincent14, 2021-03-07:
That's a good idea, i'll give a try, but i'm not sure of the result because i had a IP on my ethernet client (corresponding to the dnsmasq range)

Is it important/required to have an up to date OS ?

myxxmikeyxx, 2021-03-07:
No it’s not. If you would like you can make an img file of your SD card and I can try to install it. Is the SD card 8GB or 16GB or larger?

Vincent14, 2021-03-07:
It's only 2GB 😅

myxxmikeyxx, 2021-03-07:
Even better, do you have win32disk imager? I use that to create img backups of SD cards for the pi.

Vincent14, 2021-03-07:
No but I can use 'dd' in CLI, do you want my system?

myxxmikeyxx, 2021-03-07:
Sure. Now did you start fresh and are still getting that problem?

Vincent14, 2021-03-07:
Yes, I just get some success and fails, that's exciting! First, I would send you the .img of my current SDcard, I guess containing necessasy logs to help you to understand what happen on my hardware/network. I would send you in private message but I don't know how to do that on Github?

I've done everything from scratch (the bash history will help you to see also): format, setup wifi, install dnsmasq, curl the script, put in /etc/rc.local, restart.

AND that worked for about 5 minutes, I had the time to perform some speed tests. Suddenly, everything stopped working, and it was the same situation as previously: I have an IP, but no internet connection over Ethernet :/ I switched from cable, restarted the Pi, nothing changed :/

myxxmikeyxx, 2021-03-07:
If you don't have anything else sensitive just do a dropbox link or google drive shared link.

myxxmikeyxx, 2021-03-07:
Also nothing is really logged. So if the img works for me then I might have to have you step  through the script by copying command and running them directly and seeing what ones give an error output.

Vincent14, 2021-03-07:
Okai! I've also captured for you what appear on the Pi in /var/log/syslog when the cable is plugged in the client

Link: https://www.fromsmash.com/s_kxaEbGlb-ct
Psswd: J€Su1sFr

myxxmikeyxx, 2021-03-07:
Login is not 
pi 
raspberry

myxxmikeyxx, 2021-03-08:
I noticed two things that you should add to your /etc/rc.local
one is add sudo before the bash it should read `sudo bash <path/to/file> &`
Next is the & as I added above. This is to help prevent the script form causing other problems and lets it run in the background.
I also read that you might need to so this to the script: `sudo chmod 755 <script>` 
This looks like you have done that but just in case maybe do it again with 755

Vincent14, 2021-03-08:
I've seen your mail, sorry I've switched keyboard on AZERTY and didn't notice you'll use something else :/

Well I added sudo and & and it seems to works fine. I think sudo is overkill (because /etc/rc.local is launched by root, you can write before loading the script echo "Starting wi-to-eth-route by $(whoami)") but I use it anyway and no disconnection by now.

I guess that "&" solve something, but (but but but !) now I can see a dnsmasq error at boot that I haven't before (before it works!):
Failed to start dnsmasq
See 'systemctl status dnsmasq.service' for details

When I look into logs, it fails on "Too few arguments". 

This is important, because my setup only works if ethernet cable is plugged is AFTER boot. If I let the cable plugged in the ethernet client and just reboot the Pi, everything boot without any error AND no Internet connection on the client.

I guess that if the Pi boot without ethernet client connected to the cable, eth0 is not setup by the system, and your script can configure eth0 by itself. But if a client is ready to receive a network config, well your script isn't able to override eth0 configuration, could you check if you have this behaviour please?

I think it's something related to default configuration of dnsmasq (/etc/dnsmasq.conf) or dhcp request on eth0 done previously by the system.

Vincent14, 2021-03-08:
Ok, so:
- if the Pi boot without cable plugged-in, launch the script, then I plug a cable, I got IP+Internet on ethernet client
- if the Pi boot with the cable attached, then I got IP with no internet on ethernet client (and 'ping -I enp0s31f6 8.8.8.8' lose all packets)

I commented off the last to lines in /etc/dnsmasq.conf, no change, I'm investivating but not sure where to search :/

Vincent14, 2021-03-08:
I've tried different things without success, but I seem to have the same problem than https://github.com/arpitjindal97/raspbian-recipes/issues/43#issuecomment-633499311 (issue #43 )

Vincent14, 2021-03-08:
That's far from perfect, but I've solved my problem while editing the wifi-to-eth-route.sh script by shutting down then power up my eth0 interface https://github.com/arpitjindal97/raspbian-recipes/pull/50

I also added the README enhancement =)

myxxmikeyxx, 2021-03-08:
Well that isn’t the best fix but I would post what commands you added here. It’s the same as what it is suppose to do but if you put the script in the startup then it starts but the adaption might not get “reboot/reset” because every boot up process might not be finished. After work today I wan going to test something and hopefully have a better plan or results.

Vincent14, 2021-03-08:
Well I haven't even thank you and the contributors for this Amazing project! Thanks a lot and for your help ❤️

closed 2021-03-08 by Vincent14