arpitjindal97/raspbian-recipes #51: Cannot connect SSH to device using Raspberry internet.

KeramxD, 2021-03-26:
Thanks for wifi-to-eth-route.sh! It's amazing and works.
I shared internet from Raspberry Pi to another linux device. Internet works.
My raspberry has ip 192.168.1.xxx. Another linux device has 192.168.2.55 and I cannot connect to it via SSH from my local network. If I connect this device directly to router via ethernet it works.

What should I do to connect via SSH to Linux device that uses internet from Raspberry Pi? :)

Thank you!

myxxmikeyxx, 2021-03-26:
Can you ssh into the raspberry pi?
If so watch [this video](https://youtu.be/ySDCAJkOVOA).
Go to 7:25 in and do the ssh port forwarding.
The computer you are trying to access is technically on a different network so you need to port forward from the raspberry pi that is “on” the different network.

KeramxD, 2021-03-26:
Yes, I can ssh to Raspberry Pi without issue.
I cannot ssh into linux device that is connected to Raspberry Pi via ethernet. Cannot ssh directly from Raspberry Pi and another PC from local network.

myxxmikeyxx, 2021-03-26:
So when you are connect to the raspberry you you can’t ssh into the 192.168.2.55 IP from the raspberry pi?

KeramxD, 2021-03-26:
Correct. But when I plug this device directly to router and it get IP 192.168.1.XXX I can connect from any device so SSH on this device works. SSH does not work when device is connected to Raspberry with IP 192.168.2.XXX

myxxmikeyxx, 2021-03-26:
Strange, I will see if I can recreate the situation and get back. In the mean time someone else might be more help.

arpitjindal97, 2021-03-26:
I would like you to open two terminals

First Terminal:
```
ssh -N -T -L 2222:192.168.2.55:22 192.168.1.xxx(raspberry-ip)
```

Second Terminal:
```
ssh -p 2222 <remote-user>@localhost
```

The first one will open tunnel via Raspberry and will forward port 22 of `192.168.2.55` to your local system on port `2222`. It will not output anything on screen.

The second is basically SSHing into port `2222`. If it works then fine or else we will have to debug further

KeramxD, 2021-03-26:
I started first terminal. There was nothing for long time. When I did it one more time I see:
`marek@Air-Marek ~ % ssh -N -T -L 2222:192.168.2.55:22 pi@192.168.1.122         
pi@192.168.1.122's password: 
bind [127.0.0.1]:2222: Address already in use
channel_setup_fwd_listener_tcpip: cannot listen to port: 2222
Could not request local forwarding.`

Second terminal shows nothing.

arpitjindal97, 2021-03-26:
The first will not output anything. For the second, it should work. Looks like I will need to simulate the setup on RPi.

myxxmikeyxx, 2021-04-13:
When I ran ssh -N -T -L 2222:192.168.2.85:22 pi@192.168.1.79
it worked becuae when i did the second terminal I was able to login
`ssh -p 2222 install@localhost`

192.168.2.85  - IP of the Ubuntu Computer
install - user for Ubuntu Computer
192.168.1.79 - pi IP
pi - user for raspberry pi
2222 the port that we passed through to the local machine linked to Ubuntu machines port 22 passed through the raspberry pi

myxxmikeyxx, 2021-04-23:
@KeramxD Any update? Also after reading your post more you can change the SSh forwarding from `2222` to really any port. `2222` might already be in use on your machine for something else. Maybe try `8080`? This is used for most https websites and if 8080 was in use on your machine then surfing the net would become a lot more difficult.

Maybe try: `ssh -N -T -L 8080:192.168.2.55:22 pi@192.168.1.122`