arpitjindal97/raspbian-recipes #52: How to identify the DHCP-assigned IP address of a routed client?

neilbalch, 2021-03-26:
I'm trying to `ssh` into another device I connected to the Raspberry Pi running your [wifi-to-eth-route.sh](https://github.com/arpitjindal97/raspbian-recipes/blob/master/wifi-to-eth-route.sh) script. However, I can't seem to figure out how to identify what IP address it was given.

I tried looking in the `journalctl` output for the `systemd` service I setup to run your script on startup, but there seems to be no relevant output. Could you shed some light on this issue? Thanks!

arpitjindal97, 2021-03-27:
I think dnsmasq stores it somewhere in the file. See [this](https://stackoverflow.com/questions/51607503/how-to-list-the-assigned-leased-ip-adresses-by-dhcp-server-in-rapberry-pi)

neilbalch, 2021-03-27:
Ohan I also just realized I'm having issues with the client not receiving internet access when using the script... :)

arpitjindal97, 2021-03-28:
Try to install all the packages mentioned in the top section of script.

neilbalch, 2021-03-28:
> Try to install all the packages mentioned in the top section of script.

All of the commands are run without error, so I don't think that's the issue.

neilbalch, 2021-04-13:
> 
> 
> After running all the commands individually did it work for getting internet access?
> We had the issue where we needed bring the interfaces down then up to apply our changes but I know I had the issue where the script was running too soon so I needed to wait for everything to finish then have the script run so under #! bin/bash/ I put `sleep 120` so it waited 2 minutes to start up. Also so something about maybe putting it in `crontab -e` doing something like `@reboot sleep 10 && sudo bash /home/pi/wifi-to-eth-route.sh`

Interesting. Admittedly, I'm not very skilled or experienced with using Linux network interface configuration tools, so I don't know what useful information I can provide here to help resolve the issue.

As for the premature run, I don't think this is the problem. The script doesn't work _(i.e. provide bridged internet access on the LAN port)_ if I run it manually after logging in via `ssh` after boot.

myxxmikeyxx, 2021-04-13:
Is it possible to start form a fresh Raspbian install?
I just did a fresh install tonight and then connect to my Wi-Fi.
I enabled ssh and then used putty on my desktop
then did `sudo apt-get update && sudo apt-get upgrade -y`
let that finish then I did `sudo apt-get install dnsmasq -y`
next I did `cd /home/pi && wget https://raw.githubusercontent.com/arpitjindal97/raspbian-recipes/master/wifi-to-eth-route.sh`
after that I added it to `sudo nano /etc/xdg/lxsession/LXDE-pi/autostart`
I added `sudo bash /home/pi/wifi-to-eth-route.sh`
saved it and did a reboot
after reboot I checked by doing `ifconfig` and eth should have an IP of 192.168.2.1

neilbalch, 2021-04-13:
Okay, interesting! I probably won't have time to try this until next weekend though.

myxxmikeyxx, 2021-04-23:
@neilbalch Are you still able to test it this weekend?

neilbalch, 2021-04-23:
> 
> 
> @neilbalch Are you still able to test it this weekend?

Oh crap, lol... I forgot last weekend. I'll do my best to try this Saturday...

neilbalch, 2021-04-26:
Sorry guys, I've been very busy with school lately... I'll get to it soon. 😦

myxxmikeyxx, 2021-04-26:
@neilbalch not a problem. I am in college and have been super busy with class projects and finals around the corner so I understand the feeling well.

neilbalch, 2021-04-26:
Ah! Progress? I get this issue when running the script on a fresh and fully updated RPi: ([https://superuser.com/q/1480986/342885](https://superuser.com/q/1480986/342885))
```

pi@RPi3Bplus:~ $ bash wifi-to-eth-route.sh
iptables/1.8.2 Failed to initialize nft: Protocol not supported
iptables/1.8.2 Failed to initialize nft: Protocol not supported
iptables/1.8.2 Failed to initialize nft: Protocol not supported
iptables/1.8.2 Failed to initialize nft: Protocol not supported
iptables/1.8.2 Failed to initialize nft: Protocol not supported
pi@RPi3Bplus:~ $
```

Seems to be related to a change in iptables' API or being depricated in the future?

myxxmikeyxx, 2021-04-26:
Did you make sure to do:
`sudo apt-get update
sudo apt-get upgrade -y`

Did you also install the required packages:
`sudo apt-get install dnsmasq -y`

Also can you run this and screen shot or copy the output:
`cat /etc/os-release`

And lastly can you do:
`iptables --version`

neilbalch, 2021-04-26:
> 
> 
> Did you make sure to do:
> `sudo apt-get update sudo apt-get upgrade -y`
> 
> Did you also install the required packages:
> `sudo apt-get install dnsmasq -y`

Yes to both. :) I followed your steps exactly, but I never added the command to any autorun script. I just ran it manually from the bash prompt.

> Also can you run this and screen shot or copy the output:
> `cat /etc/os-release`
> 
> And lastly can you do:
> `iptables --version`

```
pi@RPi3Bplus:~ $  cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@RPi3Bplus:~ $ iptables --version
iptables v1.8.2 (nf_tables)
pi@RPi3Bplus:~ $
```