arpitjindal97/raspbian-recipes #53: Internet Connectivity Issue with wifi-to-wifi-route.sh

CobraBitYou, 2021-05-26:
I have used the Wi-Fi to Ethernet script flawlessly on another system and decided to try the Wi-Fi to Wi-Fi script. Unlike the Ethernet script, this one doesn't share internet connectivity. Although I can see and connect to the network, there is not internet access. Am I doing something wrong with this one?

CobraBitYou, 2021-05-26:
Hmm.... I was reading the comments from [this forum post](https://raspberrypi.stackexchange.com/questions/48307/sharing-the-pis-wifi-connection-through-the-ethernet-port). One said to uncomment the line `#conf-dir=/etc/dnsmasq.d` in the `/etc/dnsmasq.conf` file. I did that and now it shares the internet. As the user said, it doesn't recognize the `/etc/dnsmasq.d/custom-dnsmasq.conf` file until you do that. Maybe add a line into the script like this: `echo "conf-dir=/etc/dnsmasq.d" >> /etc/dnsmasq.conf` (use an if parameter to see if the line already exists).

CobraBitYou, 2021-05-26:
Actually, the page I was opening was cached on my phone. Now it won't connect at all....

arpitjindal97, 2021-06-03:
In Wifi-to-Wifi script, router is also culprit sometimes. I couldn't able to figure out the problem in this case.