arpitjindal97/raspbian-recipes #55: New recipe

onlize, 2021-08-16:
Hello, I am looking for a solution where I can use eth interface to connect to the internet, but I want to use two internal interfaces (wlan0 and eth1) for this router. Here is the idea. I am going to use eth0 as an interface that connects me to the public subnet. I will use wlan0 as AP. I also added a USB network card (eth1) and I want to connect it directly to my computer using the rj45 cable. eth1 and wlan0 should be on the same subnet.

Can I do this?

arpitjindal97, 2021-08-16:
I think it's possible, new interface `eth1` needs to added in existing script

hlastras, 2021-08-18:
Do you mean, in the script wifi-to-eth-route.sh use wlan="eth0" and eth="eth1" to share internet from eth0 to eth1?
I'm trying it this way and it doesn't work, I'm going to continue investigating.

arpitjindal97, 2021-08-18:
You will need `eth-to-wifi-route.sh` script for this task. Internet is coming in from `eth0`, `wlan0` will be used for creating Hotspot. Within this script, add some code for `eth1` from `wifi-to-eth-route.sh`.

@hlastras Yes, you are on right track.